chilipeppr/serial-port-json-server #1: Option to manually specify port

stickystyle, 2016-03-18:
At https://github.com/chilipeppr/serial-port-json-server/blob/master/seriallist_darwin.go#L39 you've set OS X based machines to only list avaialbe ports that start with "tty.". I'm trying to setup some unit tests for an application and the approach I've used in the past is to create a pair of virtual serial ports with socat like this ```$socat -d -d pty,raw,echo=0 pty,raw,echo=0```. On an OS X machine that will give me two ports that are linked together named like /dev/ttys001 & /dev/ttys002, in the past what I've done is have my application connect to /dev/ttys001 and a mocking application connect to /dev/ttys002 to emulate the hardware device. However since serial-port-json-server only enumerates at TTY's that start with "tty.", I'm not able to connect it to the virtual port that I created with socat.

So would it be feasible to add some type of command line argument to manually specify ports that are available to use? This would be different from ```-regex``` since that only filters ports that were enumerated automatically based on the "tty." pattern.

chilipeppr, 2016-03-18:
Yes, that would be very easy to do. Would you be up for the change and then
sending me a pull request?

stickystyle, 2016-03-22:
I'll give it a shot, my go skills are much more advanced than a hello world ;-)

I did try just editing the search pattern to one that would include these virtual serial ports and recompiling, and while it showed the virtual serial ports in the console during startup, they were still not available to use in serial-port-json-server. I suspect there is some type of introspection on the ports in your code to fetch some metadata about them (as I notice they get presented with a "nice" name) that these virtual ports don't provide, but I'll dig around some more.

chilipeppr, 2016-03-22:
If a nice name can't be found, the original file list name is still
included I believe. Although I might be remembering wrong how I did it.