chilipeppr/serial-port-json-server #4: Lightweight port to ESP-8266?

JamesNewton, 2016-04-02:
Just asking for any thoughts on the feasibility of supporting the ESP-8266 as a host for this interface. It would vastly reduce the cost of web enabled communications to the boards. 

If you haven't seen them, the ESP-8266 is a microscopically priced 32 bit RISC based, wifi enabled, embedded controller with ~80K of RAM and 0.5 up to 4M of FLASH. The price is... unbelievably low for the power and connectivity available. Just enough IO pins for a serial port. Perfect for this sort of application.
http://techref.massmind.org/techref/esp-8266.htm

But... I doubt this code can fit in the available program and data memory. And it seems to be bloat... I mean... growing with additional features. So the questions are:

1. How many of those commands are really necessary for Chilipeppr to operate? Will Chilipeppr stick to that list of command or demand more in the future?

2. Would it make more sense to try to add an option to this project for a lightweight, minimum to support Chilipeppr, version for embedded controllers or should that be a separate project?

chilipeppr, 2016-04-02:
Hi James, discussions started about 6 months ago on using ESP8266 on a
TinyG controller running some version of SPJS. I think it is quite
possible. There is an a NodeMCU workspace as part of ChiliPeppr that was
built to make it easy to program an ESP8266 using Lua to do just what
you're talking about. There are a couple of approaches though:

1. Use ESP8266 with websockets and have ChiliPeppr connect direct to it and
send Gcode commands.
2. Use ESP8266 with TCP socket and use SPJS as a proxy from the websocket
to the TCP socket to send/receive.
3. Use ESP8266 with MQTT and use SPJS as a proxy from the websocket to the
MQTT to send/receive.
4. Use ESP8266 with web server on it and do HTTP to and from ChiliPeppr.

The reality that SPJS is getting used more and more to connect to multiple
serial ports at the same time so you can control things like an Arduino or
other controllers that are secondary to the main controller. This is
especially true with driving lasers, or pick and place machines, or even
crazy machines like the Texterator. So the design of all of this should
take this into account.

Also, SPJS is very valuable as a re-broadcaster so that if folks have an
extra tablet acting as a DRO it shows the same data your main
laptop/desktop shows or if you are using an old Android phone as a jog
controller for your CNC you need a central re-broadcaster which is what
SPJS does.

Also, when folks add-on components like a ShuttleXpress jog device or an
Xbox controller, SPJS acts as the re-broadcaster so the browser updates
with the moves sent from those devices as well.

There was the Cayenn protocol added to the recent versions of SPJS so that
an ESP8266 device could send out a broadcast message to announce "i am
here" and SPJS would listen for it and tell all listenings sockets "hey,
there's a device by this name if anybody cares" and that would be a way for
ChiliPeppr to automatically detect your TinyG or other controller that has
an ESP8266. It works pretty well as the initial discovery mechanism, which
makes it almost nicer than serial ports.

I think it could make the most sense to leave SPJS as the re-broadcaster
and have ESP8266 connect to it and still let the browser send everything to
it, even if that means CP -> SPJS -> ESP8266 -> TinyG and then any status
updates back would be TinyG -> ESP8266 -> SPJS -> CP.

-John

JamesNewton, 2016-04-03:
LOL. There is apparently a developers forum or mailing list or something somewhere... sorry if I missed it and posted this issue instead. 

I like your option 4 the best, because then you can hit the ESP 8266 with your web browser and talk to it for debugging / configuration purposes. I'm not crazy about the idea of having to install SPJS on the PC if it could be on the ESP 8266... one less step. But any of those probably work... 

Where do y'all discuss this and how can I help? I'm probably not the right person to be doing development on this as I've never touched go and have limited exposure to ESP-8266's, but if no one else is up for it, I can try. C is an old friend, and I really think this could be worth supporting.

chilipeppr, 2016-04-03:
Well, the discussion is right here or on Google+. There hasn't been a ton
of discussion, but a decent amount. All development is open to anyone who
wants to jump in, so it would be great to have you help out. We could all
use the help.

bdurbrow, 2016-04-08:
Also, the ESP32 is on the horizon, and it is much more powerful than the 8266.