chilipeppr/serial-port-json-server #5: IPv4 or IPv6

Gustavvr, 2016-04-04:
I would like to add a CLI switch to force ipv4|6.
-ip4
-ip6

If I create a pull request will you consider it?

It looks pretty easy. In the net commands "udp" just need to be changed to "udp4" or "udp6"

chilipeppr, 2016-04-04:
I welcome any and all contributions. SPJS mainly uses websockets though
which is TCP. I just bind to a port and go, so I would almost assume that
IPv6 would work out of the box, but not sure. I do use another websocket
library (i think) but that was from like 2 yrs ago that I wrote that part
so don't totally recall.

The only part of SPJS that uses UDP is the Cayenn protocol for IoT device
discovery and communication.

Gustavvr, 2016-04-04:
It works fine with IPv6 out of the box the issue I was having is it was preferring IPv6 and I wanted it on v4.  After some testing I realized you can force it to the IP version with the -addr CLI option
-addr="[::1]:8989"
-addr="127.0.0.1:8989"

The UDP that Cayenn is not affected by this change and prefers IPv6.  This behaviour can be changed by making the changes in Cayenn.go

ServerAddr, err := net.ResolveUDPAddr("udp", ":8988")
ServerConn, err := net.ListenUDP("udp", ServerAddr)

Changing "udp" to "udp4" or "udp6" will force it one way or another.

I am not at the moment using the Cayenn and the -addr workaround is working for me so I am not going to work on a pull request at this time.  If someone really needs this PM me and I will try and find some time.

closed 2016-04-04 by Gustavvr