chilipeppr/serial-port-json-server #11: mac build?

iceboundflame, 2016-05-20:
Is there a Mac pre-built binary available? The documentation says so but I don't see it on the releases page.

chilipeppr, 2016-05-20:
Google plus page has it

Fl0p, 2016-07-28:
http://chilipeppr.com/downloads/v1.86/serial-port-json-server-1.86_darwin_amd64.zip

01ivr3, 2018-03-21:
Anyone know where there's no macOS build for v1.95? ref: https://github.com/chilipeppr/serial-port-json-server/releases

chilipeppr, 2018-03-21:
Yes, because Apple makes it impossible to let Go cross-compile from Linux
where I run the build script and I have no Apple machine to compile on. If
you could compile it and send me the binary, I'd be more than happy to post
it to the Github releases page.

houda84, 2020-04-16:
I have an access to apple machine. just let me know what I should do.  Thanks in advance.

chilipeppr, 2020-04-17:
That would be great. Maybe just email as an attachment to jlauer12@gmail.com
or post on drop box or another github repo?

houda84, 2020-04-18:
Please find below repo for mac version. Working like charm on Catalina, followed your readme file. Please updated repo location (step6) and and add a a step 6.5 if you don't have  crypto ssh type "go get -u golang.org/x/crypto/..."  .

[https://github.com/houda84/Serial_Port_JSON_Server_Mac](url)

chilipeppr, 2020-04-18:
Did you try the Darwin build in the 1.96 release? I actually forgot that Go does allow Mac cross-compiling now. https://github.com/chilipeppr/serial-port-json-server/releases

[image]

houda84, 2020-04-18:
It is executable but not functional. I get only [Last login: Sat Apr 18
21:23:48 on ttys000 ].

jaclubjly, 2020-12-06:
Hi,
When I download "serial-port-json-server-1.96_darwin_amd64.zip", I get a TexEdit document, not a Unix Executable. (Mac OS 10.15.7 Catalina)
[image]
I don't know what to do, thanks for your help.

chilipeppr, 2020-12-06:
Do a chmod 777 to give it permissions to be to an executable file.

jaclubjly, 2020-12-06:
Thank you for your quick response !
I can connect to 2 ports tty.usbmodem146201 and cu.usbmodem146201.
Both seam to work fine, is one better than the other ?