chilipeppr/serial-port-json-server #12: Error during build

dgtlmoon, 2016-05-24:
I had a few packages that were missing, so I ran `apt-get install build-essentials`

And now

```osmc@osmc:~/go$ go get github.com/chilipeppr/serial-port-json-server
# golang.org/x/crypto/ed25519
src/golang.org/x/crypto/ed25519/ed25519.go:54: undefined: crypto.SignerOpts
# github.com/gorilla/websocket
src/github.com/gorilla/websocket/client.go:361: unknown tls.Config field 'GetCertificate' in struct literal
```

This is on a debian based system on rPi (osmc distro)

chilipeppr, 2016-05-24:
build-essentials probably doesn't help the go language at all.
a. you could just use the binary for linux that's in the releases of the
github repo
b. you should do go get's for all the libraries you get errors for

dgtlmoon, 2016-05-24:
yes ofcourse :) I'm just posting the issue here for reference

dgtlmoon, 2016-05-24:
I'm running the binary without problems btw.

dgtlmoon, 2016-05-24:
So, something `go get crypto` ?

chilipeppr, 2016-05-24:
Look at the error and see the library it's referring to and that's what you
"go get"