chilipeppr/serial-port-json-server #15: Bind to specific interface

koppi, 2016-06-20:
How to make spjs bind to a spefic network interface?

My Linux system has an eth0 and a wlan0 network interface:

```bash
machinekit@CX9020:/opt/serial-port-json-server-1.92_linux_arm$ /sbin/ifconfig 
eth0      Link encap:Ethernet  HWaddr 00:01:05:24:8b:df  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0:0    Link encap:Ethernet  HWaddr 00:01:05:24:8b:df  
          inet addr:192.168.2.92  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:319 errors:0 dropped:0 overruns:0 frame:0
          TX packets:319 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:154172 (150.5 KiB)  TX bytes:154172 (150.5 KiB)

wlan0     Link encap:Ethernet  HWaddr 00:e1:b0:87:75:77  
          inet addr:192.168.0.171  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:79800 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28012 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:82418785 (78.6 MiB)  TX bytes:4266889 (4.0 MiB)
```

```bash
$ ./serial-port-json-server  -v   
2016/06/21 01:12:45 main.go:97: Version:1.92
2016/06/21 01:12:45 main.go:104: Hostname: CX9020
2016/06/21 01:12:45 main.go:110: Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect.
2016/06/21 01:12:45 seriallist_linux.go:292: Rootpath:/sys, Numscanned:8242
Matchedfiles:
/sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/manufacturer
2016/06/21 01:12:46 seriallist_linux.go:292: Rootpath:/sys, Numscanned:8242
Matchedfiles:
/sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/1-1/1-1.2/1-1.2:1.0/input/input0/id/product
/sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/product
2016/06/21 01:12:46 seriallist_linux.go:118: The list of directories with serial port device data:[/sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1 /sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/1-1/1-1.2/1-1.2:1.0/input/input0/id]
2016/06/21 01:12:46 seriallist_linux.go:131: 	Directory searching: /sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1
2016/06/21 01:12:46 seriallist_linux.go:159: This is a hub, so skipping. /sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1
2016/06/21 01:12:46 seriallist_linux.go:131: 	Directory searching: /sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/1-1/1-1.2/1-1.2:1.0/input/input0/id
2016/06/21 01:12:46 seriallist_linux.go:168: Problem reading in manufacturer text file. It does not exist or Permissions maybe? err:open /sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/1-1/1-1.2/1-1.2:1.0/input/input0/id/manufacturer: no such file or directory
2016/06/21 01:12:46 seriallist_linux.go:207:  : 0809 () DevClass:
2016/06/21 01:12:46 seriallist_linux.go:309: Rootpath:/sys/devices/platform/soc/50000000.aips/53f80200.usb/ci_hdrc.1/usb1/1-1/1-1.2/1-1.2:1.0/input/input0/id, Numscanned:5
Matcheddirs:
2016/06/21 01:12:46 seriallist_linux.go:226: 	list of ports on this. map:map[]
2016/06/21 01:12:46 seriallist_linux.go:227: 	.
2016/06/21 01:12:46 seriallist_linux.go:275: Final port list: []
2016/06/21 01:12:46 main.go:193: The Serial Port JSON Server is now running.
2016/06/21 01:12:46 main.go:194: If you are using ChiliPeppr, you may go back to it and connect to this server.
2016/06/21 01:12:46 main.go:222: Missing tls cert and/or key. Will not start HTTPS server.
2016/06/21 01:12:46 serial.go:123: Inside run of serialhub
2016/06/21 01:12:46 cayenn.go:62: UDP server running on port 8988 to listen for incoming device announcements.
2016/06/21 01:12:46 main.go:203: Starting http server and websocket on 192.168.2.92:8989
2016/06/21 01:12:46 main.go:152: Your serial ports:
2016/06/21 01:12:46 main.go:159: 	{/dev/ttyS0 ttyS0 []      }
2016/06/21 01:12:46 main.go:159: 	{/dev/ttyS1 ttyS1 []      }
2016/06/21 01:12:46 main.go:159: 	{/dev/ttyS2 ttyS2 []      }
2016/06/21 01:12:46 main.go:159: 	{/dev/ttyS3 ttyS3 []      }
2016/06/21 01:12:46 main.go:159: 	{/dev/ttyS4 ttyS4 []      }
2016/06/21 01:12:46 main.go:159: 	{/dev/ttyS5 ttyS5 []      }
```

The above binds the websocket to eth0:0, but I need the websocket to be bound to wlan0.

Any ideas?

chilipeppr, 2016-06-21:
check out the command line switches. you can specify an ip to bind to which
could achieve your goal.

chilipeppr, 2016-06-21:
serial-port-json-server -?

koppi, 2016-06-21:
ok:

```bash
machinekit@CX9020:/opt/serial-port-json-server-1.92_linux_arm$ ./serial-port-json-server -addr 192.168.0.171:8989 -saddr 192.168.0.171:8990
[...]
```
binds to ```wlan0```. And for
```bash
machinekit@CX9020:/opt/serial-port-json-server-1.92_linux_arm$ ./serial-port-json-server -addr :8989 -saddr :8990
```
I would have expected the spjs to bind to both interfaces ```eth0``` and ```wlan0```.

closed 2016-06-21 by koppi

chilipeppr, 2016-06-21:
Feel free to modify source and do pull request

Closed #15 <https://github.com/chilipeppr/serial-port-json-server/issues/15>