chilipeppr/serial-port-json-server #17: json-server terminates under MacOS X Sierra

FellMel, 2016-09-26:
After I've updated my OS to MacOS X Sierra, the json server terminates after sending a few commands to tinyg.

I suppose it's a problem with the USB communication.

On the JSon server console, I get following output:

2016/09/26 10:30:46 bufferflow_tinyg_v2.go:283: < {"sr":{"posz":0.000,"vel":0.00,"stat":3,"mpoz":0.000}}
2016/09/26 10:30:46 bufferflow_tinyg_v2.go:283: < {"qr":32}
2016/09/26 10:30:46 bufferflow_tinyg_v2.go:474: OnIncomingData() end.
1970/01/01 01:00:00 serialport.go:184: Hit end of file on serial port
fatal error: unexpected signal during runtime execution
[signal 0xb code=0x1 addr=0x203fb79f44a pc=0x4108520]

goroutine 50 [running]:
runtime.throw(0x45515e0, 0x2a)
	/usr/local/go/src/runtime/panic.go:527 +0x90 fp=0xc82004d7a8 sp=0xc82004d790
runtime.sigpanic()
	/usr/local/go/src/runtime/sigpanic_unix.go:12 +0x5a fp=0xc82004d7f8 sp=0xc82004d7a8
sync.(*Pool).Get(0x46fba20, 0x0, 0x0)
	/usr/local/go/src/sync/pool.go:101 +0x40 fp=0xc82004d848 sp=0xc82004d7f8
fmt.newPrinter(0xd0)
	/usr/local/go/src/fmt/print.go:133 +0x27 fp=0xc82004d888 sp=0xc82004d848
fmt.Sprintln(0xc82004da80, 0x1, 0x1, 0x0, 0x0)
	/usr/local/go/src/fmt/print.go:270 +0x28 fp=0xc82004d8d8 sp=0xc82004d888
log.Println(0xc82004da80, 0x1, 0x1)
	/usr/local/go/src/log/log.go:295 +0x35 fp=0xc82004d920 sp=0xc82004d8d8
main.(*serport).reader(0xc820179ea0)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serialport.go:190 +0xce4 fp=0xc82004dcb0 sp=0xc82004d920
main.spHandlerOpen(0xc8200135f5, 0x1b, 0x1c200, 0xc820013618, 0x5, 0xc820013600)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serialport.go:499 +0xc68 fp=0xc82004df70 sp=0xc82004dcb0
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1696 +0x1 fp=0xc82004df78 sp=0xc82004df70
created by main.checkCmd
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/hub.go:162 +0x670

goroutine 1 [IO wait]:
net.runtime_pollWait(0x4e211e8, 0x72, 0xc82000c150)
	/usr/local/go/src/runtime/netpoll.go:157 +0x60
net.(*pollDesc).Wait(0xc820441870, 0x72, 0x0, 0x0)
	/usr/local/go/src/net/fd_poll_runtime.go:73 +0x3a
net.(*pollDesc).WaitRead(0xc820441870, 0x0, 0x0)
	/usr/local/go/src/net/fd_poll_runtime.go:78 +0x36
net.(*netFD).accept(0xc820441810, 0x0, 0x4e212e0, 0xc82000a320)
	/usr/local/go/src/net/fd_unix.go:408 +0x27c
net.(*TCPListener).AcceptTCP(0xc82002a328, 0x40a1be0, 0x0, 0x0)
	/usr/local/go/src/net/tcpsock_posix.go:254 +0x4d
net/http.tcpKeepAliveListener.Accept(0xc82002a328, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/http/server.go:2135 +0x41
net/http.(*Server).Serve(0xc820017740, 0x4e212a8, 0xc82002a328, 0x0, 0x0)
	/usr/local/go/src/net/http/server.go:1887 +0xb3
net/http.(*Server).ListenAndServe(0xc820017740, 0x0, 0x0)
	/usr/local/go/src/net/http/server.go:1877 +0x136
net/http.ListenAndServe(0x44c7118, 0x5, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/http/server.go:1967 +0x8f
main.main()
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/main.go:169 +0xd26

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1696 +0x1

goroutine 19 [IO wait]:
net.runtime_pollWait(0x4e21128, 0x72, 0xc82000c150)
	/usr/local/go/src/runtime/netpoll.go:157 +0x60
net.(*pollDesc).Wait(0xc8201e2060, 0x72, 0x0, 0x0)
	/usr/local/go/src/net/fd_poll_runtime.go:73 +0x3a
net.(*pollDesc).WaitRead(0xc8201e2060, 0x0, 0x0)
	/usr/local/go/src/net/fd_poll_runtime.go:78 +0x36
net.(*netFD).Read(0xc8201e2000, 0xc820094400, 0x400, 0x400, 0x0, 0x4da0050, 0xc82000c150)
	/usr/local/go/src/net/fd_unix.go:232 +0x23a
net.(*conn).Read(0xc8204a8098, 0xc820094400, 0x400, 0x400, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/net.go:172 +0xe4
bufio.(*Reader).fill(0xc820340060)
	/usr/local/go/src/bufio/bufio.go:97 +0x1e9
bufio.(*Reader).Read(0xc820340060, 0xc8205f51a0, 0x2, 0x8, 0xc822030000, 0x0, 0x0)
	/usr/local/go/src/bufio/bufio.go:207 +0x260
github.com/gorilla/websocket.(*Conn).readFull(0xc8200140f0, 0xc8205f51a0, 0x2, 0x8, 0x0, 0x0)
	/Users/alden/Projects/proj82_go/src/github.com/gorilla/websocket/conn.go:546 +0xa9
github.com/gorilla/websocket.(*Conn).advanceFrame(0xc8200140f0, 0x0, 0x0, 0x0)
	/Users/alden/Projects/proj82_go/src/github.com/gorilla/websocket/conn.go:570 +0x159
github.com/gorilla/websocket.(*Conn).NextReader(0xc8200140f0, 0xc8205f5190, 0x0, 0x0, 0x0, 0x0)
	/Users/alden/Projects/proj82_go/src/github.com/gorilla/websocket/conn.go:705 +0x76
github.com/gorilla/websocket.(*Conn).ReadMessage(0xc8200140f0, 0xc820016480, 0x0, 0x0, 0x0, 0x0, 0x0)
	/Users/alden/Projects/proj82_go/src/github.com/gorilla/websocket/conn.go:768 +0x55
main.(*connection).reader(0xc8201d8130)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/conn.go:21 +0x2b
main.wsHandler(0x4de0280, 0xc8201e40b0, 0xc8203da000)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/conn.go:55 +0x2fc
net/http.HandlerFunc.ServeHTTP(0x458dda8, 0x4de0280, 0xc8201e40b0, 0xc8203da000)
	/usr/local/go/src/net/http/server.go:1422 +0x3a
net/http.(*ServeMux).ServeHTTP(0xc820012900, 0x4de0280, 0xc8201e40b0, 0xc8203da000)
	/usr/local/go/src/net/http/server.go:1699 +0x17d
net/http.serverHandler.ServeHTTP(0xc820017740, 0x4de0280, 0xc8201e40b0, 0xc8203da000)
	/usr/local/go/src/net/http/server.go:1862 +0x19e
net/http.(*conn).serve(0xc8201e4000)
	/usr/local/go/src/net/http/server.go:1361 +0xbee
created by net/http.(*Server).Serve
	/usr/local/go/src/net/http/server.go:1910 +0x3f6

goroutine 35 [select]:
main.(*hub).run(0x46fbf00)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/hub.go:51 +0x7f7
created by main.main
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/main.go:161 +0xc90

goroutine 36 [select]:
main.(*serialhub).run(0x46fbf40)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serial.go:130 +0xbff
created by main.main
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/main.go:163 +0xcb4

goroutine 20 [runnable]:
main.(*connection).writer(0xc8201d8130)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/conn.go:32 +0x60
created by main.wsHandler
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/conn.go:54 +0x2ee

goroutine 13 [chan receive]:
main.(*serport).writerBuffered(0xc820179ea0)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serialport.go:228 +0xd2
created by main.spHandlerOpen
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serialport.go:496 +0xc38

goroutine 14 [chan receive]:
main.(*serport).writerNoBuf(0xc820179ea0)
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serialport.go:263 +0xae
created by main.spHandlerOpen
	/Users/alden/Projects/proj82_go/src/serial-port-json-server/serialport.go:498 +0xc5a

Snortimer, 2016-09-29:
I have the same problem. I popped into the garage to cut a file on my CNC router, and came up against this. It will stay connected for various (short) periods of time. I can restart Serial Port JSON Server, and it'll happen again in short order.

chilipeppr, 2016-09-29:
> Try this newly compiled binary that FellMell created. He compiled it on
> the latest Mac OSX and he said it works.

Check out the Github release page for SPJS. I just uploaded the binary that
FellMell created.

Snortimer, 2016-09-29:
Awesome! I'll try it out when I get home again!

Any information about what the problem might have been or what was changed in the new version?

FellMel, 2016-09-29:
I've only build the last sources and not really testet it, yet. But a first short tests looks fine.
Maybe it depends on an update of the golang compiler or on XCode. 

I plan an intensive test at weekend.

Snortimer, 2016-09-30:
Well, whatever it was, it seems to work now!

Thanks!

FellMel, 2016-10-01:
I've tested it, too. No problems in a 2 hour job.

I think, it was an incompatible linked library, which was fixed by the fresh build.