chilipeppr/serial-port-json-server #24: serial port json server crashes randomly in osx sierra (10.12)

MrPajek, 2017-01-22:
hi 

i have a problem with my setup. everything works fine until the json server randomly shutes down ( sometimes after 10 sec., somethimes after 2 minutes) 

i covered it in a short video. would be great if you could help me.

here is the video: [(https://youtu.be/8RQ333EsdLk)]

THANK YOU

regaring the error code, here it is:

Last login: Sun Jan 22 16:14:40 on ttys000
MacBook-Pro-2:~ Matthias$ /Users/Matthias/Downloads/serial-port-json-server_macosx_v1.80/serial-port-json-server ; exit;
2017/01/22 16:16:51 main.go:88: Version:1.80
2017/01/22 16:16:51 main.go:95: Hostname: MacBook-Pro-2.home
2017/01/22 16:16:51 main.go:101: Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect.
2017/01/22 16:16:51 main.go:114: Starting server and websocket on 10.0.0.30:8989
2017/01/22 16:16:51 main.go:117: The Serial Port JSON Server is now running.
2017/01/22 16:16:51 main.go:118: If you are using ChiliPeppr, you may go back to it and connect to this server.
2017/01/22 16:16:51 seriallist_darwin.go:24: getting serial list on darwin
2017/01/22 16:16:51 seriallist_darwin.go:37: Added serial port to list:  {/dev/tty.Bluetooth-Incoming-Port tty.Bluetooth-Incoming-Port}
2017/01/22 16:16:51 seriallist_darwin.go:37: Added serial port to list:  {/dev/tty.UEMEGABOOM-LWACP tty.UEMEGABOOM-LWACP}
2017/01/22 16:16:51 seriallist_darwin.go:37: Added serial port to list:  {/dev/tty.X1T-CSRGAIA tty.X1T-CSRGAIA}
2017/01/22 16:16:51 seriallist_darwin.go:37: Added serial port to list:  {/dev/tty.X1T-CSRGAIA-1 tty.X1T-CSRGAIA-1}
2017/01/22 16:16:51 seriallist_darwin.go:37: Added serial port to list:  {/dev/tty.wchusbserial1410 tty.wchusbserial1410}
2017/01/22 16:16:51 main.go:131: Your serial ports:
2017/01/22 16:16:51 main.go:136: 	{/dev/tty.Bluetooth-Incoming-Port tty.Bluetooth-Incoming-Port}
2017/01/22 16:16:51 main.go:136: 	{/dev/tty.UEMEGABOOM-LWACP tty.UEMEGABOOM-LWACP}
2017/01/22 16:16:51 main.go:136: 	{/dev/tty.X1T-CSRGAIA tty.X1T-CSRGAIA}
2017/01/22 16:16:51 main.go:136: 	{/dev/tty.X1T-CSRGAIA-1 tty.X1T-CSRGAIA-1}
2017/01/22 16:16:51 main.go:136: 	{/dev/tty.wchusbserial1410 tty.wchusbserial1410}
2017/01/22 16:16:51 main.go:141: You can enter verbose mode to see all logging by starting with the -v command line switch.
fatal error: unexpected signal during runtime execution
[signal 0xb code=0x1 addr=0x5dd129f4d7 pc=0x40b3bb9]

goroutine 11 [running]:
runtime.gothrow(0x44712d0, 0x2a)
	/usr/local/Cellar/go/1.4.1/libexec/src/runtime/panic.go:503 +0x8e fp=0xc2080cd628 sp=0xc2080cd610
runtime.sigpanic()
	/usr/local/Cellar/go/1.4.1/libexec/src/runtime/sigpanic_unix.go:14 +0x5e fp=0xc2080cd678 sp=0xc2080cd628
sync.(*Pool).Get(0x45e9a70, 0x0, 0x0)
	/usr/local/Cellar/go/1.4.1/libexec/src/sync/pool.go:101 +0x49 fp=0xc2080cd6c8 sp=0xc2080cd678
fmt.newPrinter(0x0)
	/usr/local/Cellar/go/1.4.1/libexec/src/fmt/print.go:133 +0x2a fp=0xc2080cd708 sp=0xc2080cd6c8
fmt.Sprintf(0x440a830, 0xc, 0xc2080cd8a8, 0x1, 0x1, 0x0, 0x0)
	/usr/local/Cellar/go/1.4.1/libexec/src/fmt/print.go:202 +0x34 fp=0xc2080cd758 sp=0xc2080cd708
log.Printf(0x440a830, 0xc, 0xc2080cd8a8, 0x1, 0x1)
	/usr/local/Cellar/go/1.4.1/libexec/src/log/log.go:276 +0x50 fp=0xc2080cd7a8 sp=0xc2080cd758
main.(*BufferflowGrbl).OnIncomingData(0xc208068180, 0xc208cf88a0, 0x20)
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/bufferflow_grbl.go:110 +0x300 fp=0xc2080cdab8 sp=0xc2080cd7a8
main.(*serport).reader(0xc208042480)
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/serialport.go:92 +0x368 fp=0xc2080cdd78 sp=0xc2080cdab8
main.spHandlerOpen(0xc20803b595, 0x19, 0x1c200, 0xc20803b5b6, 0x4)
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/serialport.go:329 +0xede fp=0xc2080cdfb8 sp=0xc2080cdd78
runtime.goexit()
	/usr/local/Cellar/go/1.4.1/libexec/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc2080cdfc0 sp=0xc2080cdfb8
created by main.checkCmd
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/hub.go:151 +0x511

goroutine 1 [IO wait]:
net.(*pollDesc).Wait(0xc2080100d0, 0x72, 0x0, 0x0)
	/usr/local/Cellar/go/1.4.1/libexec/src/net/fd_poll_runtime.go:84 +0x47
net.(*pollDesc).WaitRead(0xc2080100d0, 0x0, 0x0)
	/usr/local/Cellar/go/1.4.1/libexec/src/net/fd_poll_runtime.go:89 +0x43
net.(*netFD).accept(0xc208010070, 0x0, 0x46a1ba0, 0xc2080a1760)
	/usr/local/Cellar/go/1.4.1/libexec/src/net/fd_unix.go:419 +0x40b
net.(*TCPListener).AcceptTCP(0xc208038028, 0x40ea84e, 0x0, 0x0)
	/usr/local/Cellar/go/1.4.1/libexec/src/net/tcpsock_posix.go:234 +0x4e
net/http.tcpKeepAliveListener.Accept(0xc208038028, 0x0, 0x0, 0x0, 0x0)
[... 79 lines cut ...]

goroutine 15 [chan receive]:
main.(*serport).writerBuffered(0xc208042480)
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/serialport.go:167 +0x86
created by main.spHandlerOpen
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/serialport.go:326 +0xeb4

goroutine 16 [chan receive]:
main.(*serport).writerNoBuf(0xc208042480)
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/serialport.go:196 +0x6c
created by main.spHandlerOpen
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/serialport.go:328 +0xece

goroutine 19 [chan send]:
main.func·003()
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/bufferflow_grbl.go:417 +0x50
created by main.(*BufferflowGrbl).SetPaused
	/Users/ril3y/go/src/github.com/johnlauer/serial-port-json-server/bufferflow_grbl.go:421 +0x128
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Prozess beendet]

chilipeppr, 2017-01-22:
We're on version 1.94, so my first guess would be you're on an old version.

MrPajek, 2017-01-22:
I am sorry but I can't find a newer version for OS X

chilipeppr, 2017-01-22:
https://github.com/chilipeppr/serial-port-json-server/releases

chilipeppr, 2017-01-22:
> But the newest version for OSX is 1.93
> Is that's a problem?
>
> Sorry for my stupid questIons

Try 1.93 because it's way newer than 1.80 you were using. However, if you
would be up for building the latest 1.94 and sending it to me I can add it
to the releases. There are build instructions in the README. I build SPJS
for the releases in a cloud9 workspace on an Ubuntu virtual machine because
Go has cross-compiling ability, but sadly Mac still doesn't allow virtual
machines so it's impossible to build for that without a Mac. Mac is still a
closed ecosystem and one of the downsides is it makes the open source world
not be able to support it well.

MrPajek, 2017-01-23:
thanks for your help. I couldn't find the 1.93 in the first place because I was only looking on chilipeppr directly. I am going to try to build the 1.94 for you so you can upload it - no problem ! 👍

MrPajek, 2017-01-23:
oh by the way, 1.93 works perfectly fine now!

closed 2017-01-23 by MrPajek