chilipeppr/serial-port-json-server #26: serial port commands executing intermittently

jyotsnan, 2017-03-28:
I am trying to communicate with a pump.  It is a serial port device connected via USB and is on COM4.  The drivers are provided by the pump manufacturer. It is running on Windows 8.1.  
I tried sending individual send commands as well as sendjson both on the hardware fiddle as well jsfiddle. I also notice that the data received from serial port is mixed up with the transmit echo from SPJS.  Is there any way to isolate the transmit and receive communication?

Here are two Serial Port Console logs for the same sendjson commands.
1.{"P":"COM4","Data":[{"Id":"001","D":"\n"},{"Id":"002","D":"S\n"},{"Id":"003","D":"T\n"}]}
 {"P":"COM4","Data
WHAT?
COMMAND ? ":[{"Id":"001","D":"
WHAT?
COMMAND ? \n"},{"Id":"002","D"
WHAT?
COMMAND ? :"S\n"},{"Id":"003",
WHAT?
COMMAND ? "D":"T\n"}]}

2.{"P":"COM4","Data":[{"Id":"001","D":"\n"},{"Id":"002","D":"S\n"},{"Id":"003","D":"T\n"}]}
 {"P":"COM4","Data":
WHAT?
COMMAND ? [{"Id":"001","D":"\n
WHAT?
COMMAND ? "},{"Id":"002","D":"
WHAT?
COMMAND ? S\n"},{"Id":"003","D
STATUS
FORWARD DIRECTION
VOLUME: 111
PUMP READY
COMMAND ? ":"T\n"}]}
Here is a communication log with Putty for the same commands

WHAT?
COMMAND ? S
STATUS
FORWARD DIRECTION
VOLUME: 111
PUMP READY
COMMAND ? T
CURRENT EMPTY/PRIME TIME (SECONDS): 20
NEW EMPTY/PRIME TIME?

The first command which is just a newline should get a reply with What and Command strings.
Basically any command that the pump does not understand gets a reply with What and Command strings. The S command should get the Status and T the time for priming.  
I am not clear why the S command got a Status reply only once.

Here is a log of just sending a newline which in Putty replies with What and Command each time but in SPJS this happens intermittently as shown here.

 send COM4 ""
send COM4 ""
 send COM4 ""
send CO
WHAT?
COMMAND ? M4 ""
 send COM4 ""
send COM4 ""
 send COM4 ""
sen
WHAT?
COMMAND ? d COM4 ""
 send COM4 ""
send COM4 "
WHAT?
COMMAND ? "
 send COM4 ""
send COM4 ""
 send COM4 ""
send CO
WHAT?
COMMAND ? M4 ""
 send COM4 ""
send COM4 ""
 send COM4 ""
sen
WHAT?
COMMAND ? d COM4 ""

I was trying to setup a sniffer software and tried Free Serial Port Monitor and Portmon and haven't been successful in getting it to monitor the communication.
Please suggest how I could get it to work consistently.
Thanks in advance.

jyotsnan, 2017-03-31:
I figured out the issue.  The problem seems to be on the pump side.  It works fine when I send a carriage return and line feed.  I am still not sure why it responded intermittently when only a newline is sent.

Thanks again for developing this, it's extremely useful!

closed 2017-03-31 by jyotsnan

chilipeppr, 2017-03-31:
SPJS does send back character by character, but I believe the web interface
in ChiliPeppr only updates newline by newline, so you should always send
back newlines from your serial device if you are wanting to use standard
workspace in ChiliPeppr. If you custom code you can see character by
character. Not completely certain though.