chilipeppr/serial-port-json-server #34: Data arriving out of order

DanielJoyce, 2017-04-25:
Investigating the use of SPJS for talking to serial port RFID/Barcode readers. The default buffer flow is causing data to arrive out order, leading to jumbled scans. 

The reader works fine when talking to cutecom over its serial port. So something is happening in SPJS somewhere.

chilipeppr, 2017-04-25:
That is network congestion.  You need to solve that in a higher layer like
how chilipeppr does it by looking at per line id's

DanielJoyce, 2017-04-25:
Its purely localhost so network congestion is impossible.

DanielJoyce, 2017-04-25:
Hmm, I modified the timed buffer flow to increase the timeout to 500 millis and used that. No corruption. This looks like some kind of race condition somewhere. I don't think its network traffic issue as I am on localhost, and WS runs over TCP which enforces single arrival and sequencing.

DanielJoyce, 2017-04-25:
Just confirmed data is garbled on GO side.

Problem appears to be here: Lines 157-175 of serialport.go

```  go
			if p.bufferwatcher.IsBufferGloballySendingBackIncomingData() == false {
				//m := SpPortMessage{"Alice", "Hello"}
				m := SpPortMessage{p.portConf.Name, data}
				//log.Print("The m obj struct is:")
				//log.Print(m)

				//b, err := json.MarshalIndent(m, "", "\t")
				b, err := json.Marshal(m)
				if err != nil {
					log.Println(err)
					h.broadcastSys <- []byte("Error creating json on " + p.portConf.Name + " " +
						err.Error() + " The data we were trying to convert is: " + string(ch[:n]))
					break
				}
				log.Print("Printing out json byte data...")
				log.Print(string(b))
				h.broadcastSys <- b
				//h.broadcastSys <- []byte("{ \"p\" : \"" + p.portConf.Name + "\", \"d\": \"" + string(ch[:n]) + "\" }\n")
			}
```
commenting out the surrounding if test fixes the issue for the case where the default buffer protocol is used. 

```  go
//			if p.bufferwatcher.IsBufferGloballySendingBackIncomingData() == false {
				//m := SpPortMessage{"Alice", "Hello"}
				m := SpPortMessage{p.portConf.Name, data}
				//log.Print("The m obj struct is:")
				//log.Print(m)

				//b, err := json.MarshalIndent(m, "", "\t")
				b, err := json.Marshal(m)
				if err != nil {
					log.Println(err)
					h.broadcastSys <- []byte("Error creating json on " + p.portConf.Name + " " +
						err.Error() + " The data we were trying to convert is: " + string(ch[:n]))
					break
				}
				log.Print("Printing out json byte data...")
				log.Print(string(b))
				h.broadcastSys <- b
				//h.broadcastSys <- []byte("{ \"p\" : \"" + p.portConf.Name + "\", \"d\": \"" + string(ch[:n]) + "\" }\n")
//			}
```

DanielJoyce, 2017-04-25:
This seems weird to me because the default buffer protocol has that test set to false, so the body should be working...

DanielJoyce, 2017-04-25:
Ok that does seem to seriously improve things in the default bufferflow case. Logging the data looks correct. 

Still seeing weird data client side, but it may be the websocket wrapping lib I am using.

DanielJoyce, 2017-04-25:
Okay, pulled out WS lib, went to raw websocket, looks like I am getting sent the trailing piece of the last message plus a little garbage.

As this comes inside a json {"P":"","D":""} object, this is obviously being sent by SPJS from somehwere

DanielJoyce, 2017-04-25:
Weird, don't know what to believe anymore. Now its working all the time now. I even backed out my changes and it working fine now. I don't know if spjs or serial port somehow got into a bad state?

DanielJoyce, 2017-04-25:
Went and did a full clean and rebuild, and even with none of the changes above, still works fine...

closed 2017-04-25 by DanielJoyce