chilipeppr/serial-port-json-server #37: Different serial libraries used in multiple places in the code

DanielJoyce, 2017-04-27:
Which should be userd?

For example, seriallist.go has the following

``` go
import (
	//"fmt"
	//"github.com/lxn/win"
	//"github.com/mattn/go-ole"
	//"github.com/mattn/go-ole/oleutil"
	"github.com/go-ole/go-ole"
	"github.com/go-ole/go-ole/oleutil"
	//"github.com/tarm/goserial"
	//"github.com/johnlauer/goserial"
	"log"
	"os"
	"strings"

	"github.com/facchinm/go-serial"
	//"encoding/binary"
	"strconv"
	"sync"
	//"syscall"
	"regexp"
)
```

while serialport.go has
``` go
import (
	"bytes"
	"encoding/json"
	"encoding/base64"
	"sync"

	"github.com/johnlauer/goserial"
	//"github.com/facchinm/go-serial"
	"io"
	"log"
	"strconv"
	"strings"
	"time"
)
```

Which is it, `"github.com/johnlauer/goserial"` or `"github.com/facchinm/go-serial"` ?

chilipeppr, 2017-04-27:
Probably John Lauer one

DanielJoyce, 2017-04-27:
Neither apply cleanly in all cases. :P It seems feature sets have diverged and it also seems unwise to be using two serial port libraries.

DanielJoyce, 2017-04-27:
When I try and replace all occurences of serial lib with johnlauer/goserial I get the following build error

``` terminal
./programmer.go:373: undefined: serial.Mode
./programmer.go:378: too many arguments in call to serial.OpenPort
        have (string, <T>)
        want (*serial.Config)
./programmer.go:400: undefined: serial.GetPortsList
./programmer.go:404: undefined: serial.GetPortsList
./programmer.go:438: undefined: serial.Mode
./programmer.go:443: too many arguments in call to serial.OpenPort
        have (string, <T>)
        want (*serial.Config)
./seriallist.go:32: undefined: serial.GetPortsList
```

chilipeppr, 2017-04-27:
i'd leave it alone then. serial libraries serve different purposes. one
purpose is a library that has better serial port listing capabilities.
another purpose is the raw opening of the port.