chilipeppr/serial-port-json-server #39: serial-port-json-server as a service

ciro-maciel, 2017-10-29:
Hi @chilipeppr, nice your project, greatly simplifies the process of communication!

I'm thinking of using serial-port-json-server as a Windows/Linux/Mac OS as a service, you figure out what the best way to do it?

I started the tests in Windows, the service arrives until the serial-port-json-server process is started, but after a while I get the message:

`the service did not respond to the start or control request in a timely fashion`

And the process is closed.

Thank you

Ciro

chilipeppr, 2017-10-29:
What did you create the Windows service with?

ciro-maciel, 2017-10-29:
with (Inno Setup)[http://www.jrsoftware.org/isinfo.php]

and the script below:

```
; Script generated by the Inno Setup Script Wizard.
; SEE THE DOCUMENTATION FOR DETAILS ON CREATING INNO SETUP SCRIPT FILES!

#define MyAppName "My Program"
#define MyAppVersion "1.5"
#define MyAppPublisher "My Company, Inc."
#define MyAppURL "http://www.example.com/"
#define MyAppExeName "serial-port-json-server.exe"

[Setup]
; NOTE: The value of AppId uniquely identifies this application.
; Do not use the same AppId value in installers for other applications.
; (To generate a new GUID, click Tools | Generate GUID inside the IDE.)
AppId={{260BE378-BA2F-4A6C-8684-3BFB6894D261}
AppName={#MyAppName}
AppVersion={#MyAppVersion}
;AppVerName={#MyAppName} {#MyAppVersion}
AppPublisher={#MyAppPublisher}
AppPublisherURL={#MyAppURL}
AppSupportURL={#MyAppURL}
AppUpdatesURL={#MyAppURL}
DefaultDirName={pf}\{#MyAppName}
DisableDirPage=yes
DefaultGroupName={#MyAppName}
DisableProgramGroupPage=yes
OutputDir=C:\Users\ciro.maciel\Downloads\ino
OutputBaseFilename=setup
Compression=lzma
SolidCompression=yes

[Languages]
Name: "english"; MessagesFile: "compiler:Default.isl"

[Files]
Source: "C:\Users\ciro.maciel\Downloads\ino\serial-port-json-server.exe"; DestDir: "{app}"; Flags: ignoreversion
; NOTE: Don't use "Flags: ignoreversion" on any shared system files

[Icons]
Name: "{group}\{#MyAppName}"; Filename: "{app}\{#MyAppExeName}"

[run]
Filename: {sys}\sc.exe; Parameters: "create mysrv2 start= auto binPath= ""{app}\{#MyAppExeName}""" ; Flags: runhidden

[UninstallRun]
Filename: {sys}\sc.exe; Parameters: "stop mysrv2" ; Flags: runhidden
Filename: {sys}\sc.exe; Parameters: "delete mysrv2" ; Flags: runhidden

```

Do you suggest or have a different/better approach?

ciro-maciel, 2017-10-30:
Sorry for the inconvenience @chilipeppr,

I studied it. The best way and use the [NSSM] (https://nssm.cc/) in line command to create the service. Everything is running, okay!

Thank you.

Ciro

chilipeppr, 2017-10-30:
Cool. Anyway to post a binary or installer so I can share with all other
users?

ciro-maciel, 2017-10-30:
Sure!

As soon as I finish the installer, put all the steps and softwares.

Ciro

Simskii, 2018-12-20:
Did you solve this? I need an installer for the exe file and install it as a service.