chilipeppr/serial-port-json-server #41: External IP printed on log output even with -addr provided

james-coder, 2017-12-29:
This is a strange error that seems to do the wrong thing, but after examining the code, it only logs the wrong thing and ChiliPeppr is still functioning correctly.
```
C:\Users\MyUser\Downloads\serial-port-json-server-1.95_windows_amd64>serial-port-json-server.exe -addr 127.0.0.1:8888 -disablecayenn -regex COM4
2017/12/29 14:00:52 main.go:112: Version:1.95
2017/12/29 14:00:52 main.go:119: Hostname: windows-host
2017/12/29 14:00:52 main.go:125: Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect.
2017/12/29 14:00:52 main.go:146: You specified a serial port regular expression filter: COM4
2017/12/29 14:00:52 main.go:152: You can enter verbose mode to see all logging by starting with the -v command line switch.
2017/12/29 14:00:52 seriallist.go:53: serial port did not match. port: {CNCA0 CNCA0 []      }
2017/12/29 14:00:52 seriallist.go:53: serial port did not match. port: {CNCB0 CNCB0 []      }
2017/12/29 14:00:52 seriallist.go:53: serial port did not match. port: {COM3 COM3 []      }
2017/12/29 14:00:53 seriallist_windows.go:198: DeviceId elements:[ROOT\COM0COM\0001]
2017/12/29 14:00:53 seriallist_windows.go:198: DeviceId elements:[COM0COM\PORT\CNCA1]
2017/12/29 14:00:53 seriallist_windows.go:198: DeviceId elements:[COM0COM\PORT\CNCB1]
2017/12/29 14:00:53 seriallist_windows.go:231: index:0, name:COM# <-> COM#, friendly:com0com - bus for serial port pair emulator 1 (COM# <-> COM#)
2017/12/29 14:00:53 seriallist_windows.go:232: pid:, vid:
2017/12/29 14:00:53 seriallist_windows.go:231: index:1, name:COM4, friendly:com0com - serial port emulator (COM4)
2017/12/29 14:00:53 seriallist_windows.go:232: pid:, vid:
2017/12/29 14:00:53 seriallist_windows.go:231: index:2, name:COM3, friendly:com0com - serial port emulator (COM3)
2017/12/29 14:00:53 seriallist_windows.go:232: pid:, vid:
2017/12/29 14:00:53 main.go:204: Disabled loading of Cayenn TCP/UDP server on port 8988
2017/12/29 14:00:53 main.go:219: The Serial Port JSON Server is now running.
2017/12/29 14:00:53 main.go:235: Starting http server and websocket on 192.168.1.55127.0.0.1:8005
2017/12/29 14:00:53 main.go:255: Missing tls cert and/or key. Will not start HTTPS server.
2017/12/29 14:00:53 serial.go:123: Inside run of serialhub
2017/12/29 14:00:53 main.go:220: If you are using ChiliPeppr, you may go back to it and connect to this server.
2017/12/29 14:00:53 conn.go:45: Started a new websocket handler
2017/12/29 14:00:53 hub.go:122: open COM5 9600
2017/12/29 14:00:53 serialport.go:373: Inside spHandler
2017/12/29 14:00:53 serialport.go:389: Opening serial port COM5 at 9600 baud
2017/12/29 14:00:53 serialport.go:434: Just tried to open port
2017/12/29 14:00:53 serialport.go:437: Error opening port The system cannot find the file specified.
2017/12/29 14:00:54 main.go:169: Your serial ports:
2017/12/29 14:00:54 main.go:176:        {COM4 com0com - serial port emulator (COM4) [] COM0COM\PORT\CNCA1  Vyacheslav Frolov   }
```

My external IP is 192.168.1.55. However, I am asking only to bind to 127.0.0.1:8005 using the -addr switch. After reading the code and examining what is actually bound to, it appears to be functionally working, just strange logging.

**Got output:**
```
2017/12/29 14:00:53 main.go:235: Starting http server and websocket on 192.168.1.55127.0.0.1:8005
```

**Expected to see:**
```
2017/12/29 14:00:53 main.go:235: Starting http server and websocket on 127.0.0.1:8005
```

https://github.com/chilipeppr/serial-port-json-server/blob/master/main.go#L267
externalIP() doesn't seem to even consider *addr or flag.Lookup("addr")?
https://github.com/chilipeppr/serial-port-json-server/blob/master/main.go#L288
"addr" in this case is local function scope from addrs?

https://github.com/chilipeppr/serial-port-json-server/blob/master/main.go#L236
startHttp() correctly binds to *addr from line 27 of main.go#L27

https://github.com/chilipeppr/serial-port-json-server/blob/master/main.go#L235
log seems to print external IP and passed -addr option? (But doesn't attempt to de-duplicate or have logic to print correct IP?)

I'm not sure why externalIP() and flag.Lookup("addr") are concattenated in a string? Maybe just output only flag.Lookup("addr")? (Assuming it will correctly fail and die if it can't be bound to? Maybe list externalIP() output on a separate line like "IPs found:" or something?)

chilipeppr, 2017-12-29:
> master/main.go#L267
> master/main.go#L288
> master/main.go#L236
> master/main.go#L235

I think it just binds to ALL addresses by default and that I don't have
code in there to specifically bind to only one IP address. The logging is
trying to find all your IPs on it's own rather than getting them from the
TCP/IP library that does the actually binding. It was just the way Go
allowed me to do it. So, you're right. The logging is not as accurate as
the actual behavior.

james-coder, 2017-12-30:
Which is the better thing to do?

1. Log the same output as the user provides after the -addr switch? `flag.Lookup("addr")`
2. (optional) also log externalIP() on a new line?

I think I get the reason for externalIP(), it's when the string ":8989" is passed into `http.ListenAndServe(*addr...` but we actually want to tell the user what we bound to. (?) Maybe there's a different way to ask "net/http" what we bound to? https://golang.org/pkg/net/http/#ListenAndServe (edit: After looking at docs for a while, I don't see an easy way to grab the IPs/ports that the library was bound to....?)

james-coder, 2017-12-30:
I don't think there is any way to ask "net/http" what we bound to. I searched as far into golang as I could (IANAGE, GE = Go Expert). So I went searching on github on what other people do for `http.ListenAndServe()`.

Maybe be explicit once you have found the right thing with externalIP() or somehow default to blank string for IP, but display that?
https://github.com/Digivance/mvcrouter/blob/c61c7e3f1b8862ef681ca7916d5d92d48d6f6a0a/application.go#L46
`http.ListenAndServe(fmt.Sprintf("%s:%d", app.BindAddress, app.HTTPPort), handler)`

Also, I could see not wanting to break reverse compatibility with what already works, so maybe only changing what is logged is the best bet?

I'll defer to the experts.

chilipeppr, 2017-12-30:
> d48d6f6a0a/application.go#L46

I did write that block of code about 4 years ago, and Go has had many new
version releases. I would've thought they'd finally expose what IP's it's
listening to at this point. If they don't, which sounds like from your
searching they still don't, then better parsing the command line might be a
way to go, or looping through all external IP's and see if there's a
listener on the port to get a filtered/better list? Not sure, but I'm game
if you want to make any mods to that logging to make it more accurate.