chilipeppr/serial-port-json-server #51: Send commands from RPI

ameennihad, 2018-05-21:
I have a typical setup of tinyg and Rasberry PI running SPJS, my computer is few meters away from my CNC machine, sometimes I need to adjust XTZ and zero the machine when I'm near it without going beck and forth to the computer, what is the possibility of doing that from keyboard or game controller connected to RPI? Or possibly few buttons connected to GPIO pins?

I'm thinking of writing Python script to send command to SPJS.

Any better ideas?

chilipeppr, 2018-05-21:
Watch my video on YouTube of esp32 joystick for chilipeppr. It  connects in
over websocket to send additional commands in.

chilipeppr, 2018-05-23:
> A side from analog input of the joystick, is it possible to run the same
> code on ESP8266?

Yes, it would be, but I find that the cost of ESP32 is the same as ESP8266
now and the RAM limitations of ESP8266 vs ESP32 aren't worth the pain and
suffering of ESP8266 anymore.

ameennihad, 2018-05-23:
I'm considering ESP8266 because I have number of unused boards, but no ESP32.

ameennihad, 2018-06-04:
I ended up writing the code in c++, I have a functional code now, I can control the machine with a gamepad connected to Raspberry PI and do other stuff like zero-out the machine. I'm considering the following features:
- Memic the function of touch plate.
- Connect E-Stop to RPi GPIO and send feedhold when the switch is pressed, I had issues connecting the switch directly to tinyg.

I have a questions:
- Is it possible to tell SPJS to run the loaded gcode in CP? I guess no!

chilipeppr, 2018-06-04:
Yes, there would be a way to do that. There is a "broadcast" command in
SPJS that all connected websockets get copies of. So just send "broadcast
start-job" or something like that and then write a macro that auto-loads in
CP that watches for that broadcast command and sends the pubsub to play the
Gcode. Look at the pubsub docs in the Gcode widget to see the docs on the
pubsub command that plays the Gcode. It's something like
chilipeppr.publish("/com-chilipeppr-gcode-widget/play", "");

chilipeppr, 2018-06-04:
Try this macro. May have bugs. Not tested.

// This macro shows how to watch for the chilipeppr
// broadcast command, respond to the broadcast payload
// you're looking for, and then send an automatic play command
// to send your Gcode to your machine
var myWatchChiliPepprBroadcast = {
    init: function() {
        // Uninit previous runs to unsubscribe correctly, i.e.
        // so we don't subscribe 100's of times each time we modify
        // and run this macro
        if (window["myWatchChiliPepprBroadcast"]) {
            macro.status("This macro was run before. Cleaning up...");
            window["myWatchChiliPepprBroadcast"].uninit();
        }
        macro.status("Subscribing to broadcast pubsub event");

        // store macro in window object so we have it next time thru
        window["myWatchChiliPepprBroadcast"] = this;

        this.setupSubscribe();
    },
    uninit: function() {
        macro.status("Uninitting broadcast macro.");
        this.unsetupSubscribe();
    },
    setupSubscribe: function() {
        // Subscribe to both events because you will not
        // get onComplete if the controller is sophisticated
        // enough to send onExecute, i.e. TinyG will only
        // get onExecute events while Grbl will only get
        // onComplete events
        chilipeppr.subscribe("/com-chilipeppr-widget-serialport/onBroadcast",
this, this.onChiliPepprBroadcast);
    },
    unsetupSubscribe: function() {
        chilipeppr.unsubscribe(
"/com-chilipeppr-widget-serialport/onBroadcast", this.onChiliPepprBroadcast
);
    },
    onChiliPepprBroadcast: function(data) {
        macro.status("Got onChiliPepprBroadcast.");
console.log("got onChiliPepprBroadcast. data:", data);
// See if it's our "start-job" command?
if ("Cmd" in data && data.Cmd && data.Cmd == "Broadcast" && data.Msg ==
"start-job") {
chilipeppr.publish("/com-chilipeppr-elem-flashmsg/flashmsg", "Playing Gcode",
"We got a broadcast command of start-job so playing Gcode.", 3000);
setTimeout(this.playGcode, 1000);
}
    },
    playGcode: function() {
        macro.status("Playing gcode.");
        chilipeppr.publish("/com-chilipeppr-widget-gcode/play", "");
    },
    pauseGcode: function() {
        macro.status("Just paused gcode.");
        chilipeppr.publish("/com-chilipeppr-widget-gcode/pause", "");
    }
}
myWatchChiliPepprBroadcast.init();

ameennihad, 2018-06-05:
The "data" object passed to onChiliPepprBroadcast function is the message "start-job", there is no data.Cmd or data.Msg both are undefined! I just changed the if statement to following to make it work:
if (data == "start-job")
I'm sending the command this way: ws->send("broadcast start-job");
Am I missing something or this is a bug in your macro?

chilipeppr, 2018-06-05:
Bug in my macro. I didn't test it. Your change sounds perfect.

chilipeppr, 2018-06-05:
BTW, have you looked at the C code you can download for the ShuttleXpress
widget? It does pretty much what it sounds like you wrote as well.