chilipeppr/serial-port-json-server #52: Raspberry Pi3, TinyG

Seb66, 2018-06-01:
Hi I'm using a Pi3 and and Tiny G#

Im greeted by this black screen with this black screen? what am i doing wrong. 

[image]

chilipeppr, 2018-06-01:
webgl is hard to get working on a raspi. it's black cuz webgl can't
instantiate. there are other workspaces like the "tablet" workspace that
don't load the 3d viewer.

Seb66, 2018-06-01:
Hi thanks for your reply!
OK so if i load that instead how do i know how the part cuts/is at?

Sorry for asking silly questions! i just dont get the SW as im 100% hardware!

chilipeppr, 2018-06-01:
Run your browser on a PC or mac