chilipeppr/serial-port-json-server #58: Question

Taillaert, 2018-10-02:
We are using the 1.95 version of the serial json service. Is there a command to check if a comport is already open? When we sending JSON command to open com port in several browser instances we get open com3 9600 (for example) and then the JSON services stops. doesn't respond anymore to request and we need to restart

chilipeppr, 2018-10-02:
You can list the serial ports and it returns whether the port is open or
not.

I have seen the bug you're describing occur every now and then and it's
probably not hard to track down if you wanted to help fix it and check in
your changes on Github.

Taillaert, 2018-10-02:
Thank you for the reply. I've found! Seems when opening the port 2-3 times it gives this bug.

leonua, 2018-10-09:
Have problems with pushing it to git now, but looks like issue is solved with this patch:

 main.go       | 4 ++--
 serialport.go | 3 ++-
 2 files changed, 4 insertions(+), 3 deletions(-)

diff --git a/main.go b/main.go
index 6bb8eb7..e6c382a 100755
--- a/main.go
+++ b/main.go
@@ -22,8 +22,8 @@ import (
 )
 
 var (
-	version      = "1.95"
-	versionFloat = float32(1.95)
+	version      = "1.96"
+	versionFloat = float32(1.96)
 	addr         = flag.String("addr", ":8989", "http service address. example :8800 to run on port 8800, example 10.0.0.2:9000 to run on specific IP address and port, example 10.0.0.2 to run on specific IP address")
 	//	addr  = flag.String("addr", ":8980", "http service address. example :8800 to run on port 8800, example 10.0.0.2:9000 to run on specific IP address and port, example 10.0.0.2 to run on specific IP address")
 	saddr = flag.String("saddr", ":8990", "https service address. example :8801 to run https on port 8801")
diff --git a/serialport.go b/serialport.go
index ccfb160..b1467f2 100755
--- a/serialport.go
+++ b/serialport.go
@@ -437,7 +437,8 @@ func spHandlerOpen(portname string, baud int, buftype string, isSecondary bool)
 		log.Print("Error opening port " + err.Error())
 		//h.broadcastSys <- []byte("Error opening port. " + err.Error())
 		h.broadcastSys <- []byte("{\"Cmd\":\"OpenFail\",\"Desc\":\"Error opening port. " + err.Error() + "\",\"Port\":\"" + conf.Name + "\",\"Baud\":" + strconv.Itoa(conf.Baud) + "}")
-
+    spIsOpening = false
+	  spmutex.Unlock()
 		return
 	}
 	log.Print("Opened port successfully")

leonua, 2018-10-09:
[patch.win-open-open.txt](https://github.com/chilipeppr/serial-port-json-server/files/2460419/patch.win-open-open.txt)