chilipeppr/serial-port-json-server #60: Custom/3rd Party Arduino boards?

murilopolese, 2018-11-25:
Hello! First of all, this project is simply awesome! Huge thanks for everyone involved! :1st_place_medal:  I'm really excited not only about its idea and implementation (in Go!) but how it fills a gap between browser and hardware. I have been looking around for such a software/project/bridge and SPJS is brilliant! Specially after the `avrdude` addition.

Is it possible to add custom hardware just like in the Arduino IDE (as described [here](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification))? Either by manually placing files or recompiling the binaries?

Thank you very much!

chilipeppr, 2018-11-25:
> everyone involved! 🥇 I'm really excited not only about its idea and

Feel free to fork the code, modify to add custom hardware, and then send
pull requests! Thanks for the kudos on the project.

murilopolese, 2018-11-25:
Hey, @chilipeppr !

Thanks a lot for the quick reply! I'll definitely give it a Go ;) (pun not intended) 

I'm very new to the project so it might take me a while to figure out where to put all the hardware files I would normally drop into the Arduino sketch `hardware` folder: I am planning to add support to this beauty [here](https://github.com/Quirkbot/QuirkbotArduinoHardware). 

Would you have a word of advice on that?

chilipeppr, 2018-11-25:
In terms of the sketch folder, the programmer part of SPJS is just the open
source Arduino project, so it's identical to its structure. Will you have
your own USB vid/pid?

chilipeppr, 2018-11-25:
> I see... So basically the programmer is what's on the arduino submodule,
> right? Or is it related to the arduino create agent
> <https://github.com/arduino/arduino-create-agent>?
>
> Yes, I added the board definitions
> to the boards.txt inside the arduino folder but I'm getting the following
> error:
>
> program /dev/ttyACM0 arduino:avr:quirkbot /home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/blinky.hex
> {"Cmd":"-C/home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/arduino/tools/avrdude//etc/avrdude.conf -patmega32u4 -cavr109 -P/dev/ttyACM0 -b57600 -D -Uflash:w:/home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/blinky.hex:i -v","Flasher":"/home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/arduino/tools/avrdude/bin/avrdude","IsFound":"true","ProgrammerStatus":"CommandReady"}
> {"Cmd":"-C/home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/arduino/tools/avrdude//etc/avrdude.conf -patmega32u4 -cavr109 -P/dev/ttyACM0 -b57600 -D -Uflash:w:/home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/blinky.hex:i -v","ProgrammerStatus":"Starting"}
> Could not program the board
> {"Msg":"Could not program the board.","Output":"/home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/arduino/tools/avrdude/bin/avrdude: error while loading shared libraries: /home/murilopolese/Apps/serial-port-json-server-1.95_linux_amd64/arduino/tools/avrdude/bin/../lib/libusb-1.0.so.0: file too short\n","ProgrammerStatus":"Error"}
>
> I'm trying to figure out if that has to do with the avrdude or if it's
> missing something from the hardware specification (core, variants,
> libraries, bootloader, etc)

The Arduino create agent is just SPJS under the covers. The Arduino team
forked SPJS a couple years ago to create Arduino Create Agent (which was a
proud moment for SPJS). So, you can borrow code from that if it helps, but
they have made significant mods.

The other advice is you should be able to run avrdude from the command line
inside the SPJS directory first to test if what you're doing should work.
Ultimately SPJS is just running a command line from the Go code for the
programmer. I recall trying to get builds of avrdude for each platform that
had all shared libs statically linked to not run into the problem you're
seeing.

murilopolese, 2018-11-25:
Yeah! I saw it and this is awesome. Again, congrats :tada: ! SPJS is indeed a great project! I hope I can level up to be able to contribute to it :)

Hmmm... I see the issue now...  I was trying to run the binaries from the `arduino` folder when I realised I had the `serial-port-json-server-1.95_linux_amd64`. I downloaded the `serial-port-json-server-1.95_linux_386` and it managed to flash the board correctly.

But that is strange as my `lscpu` outputs:

```
lscpu
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
...
```

I also confirmed from GUI, I'm running Ubuntu 18.04.1, OS type 64-bit. Perhaps a bug or do you thing it could be related to my setup? Should I open a separated issue?

Anyway, I managed to append the `boards.txt` from the `arduino` submodule and it did recognized the new board definition. But I guess that would be a PR outsite of SPJS scope.