chilipeppr/serial-port-json-server #62: Infinite Cayenne discover/announce loop at startup

mgkeeley, 2018-12-22:
I get a never ending list of `"Announce":"i-am-your-server"` messages when connecting Chilipeppr to an already-running SPJS running on a raspberry pi.
This only happens when SPJS is already running, and I refresh the Chilipeppr browser page.
(If I stop SPJS, refresh the page, start SPJS, then connect via clicking the ws: link on the "Your Servers" recent server list, then tick the checkbox in the "Port List" it works OK).

Seems to be Chilipeppr sends a message `msg: {"Cayenn":"Discover"}`, then SPJS does a UDP broadcast `{"Cayenn":"Discover"}`. However, SPJS gets the broadcast back (is that normal?), and sends the announce back to itself. But that fails, and then it loops forever.

See the SPJS log attached.
[spjs.log](https://github.com/chilipeppr/serial-port-json-server/files/2704947/spjs.log)

Maybe SPJS should ignore Cayenn messages from itself?

chilipeppr, 2018-12-22:
We're at version 1.95 and that looks like you're running 1.88. I believe
this bug was fixed a long time ago.

mgkeeley, 2018-12-23:
https://github.com/chilipeppr/serial-port-json-server/pull/63

chilipeppr, 2018-12-23:
Awesome. Thanks for the help.

closed 2018-12-31 by mgkeeley