chilipeppr/serial-port-json-server #64: Can't run in OSX Catalina.

pastprimitive, 2019-10-10:
Can't get the 1.96 or 1.86 Darwin version to run in OSX Catalina. Just says cannot run binary.

chilipeppr, 2019-10-10:
Could you try checking out the source and building it? I don't have a Mac
and have never found a way to run a virtual machine to build for Mac.

pastprimitive, 2019-12-10:
Apologies for the long delay. I was able to get around to looking into things more in depth. It was an issue with Catalina's heightened security protocols. In order to run it. I had to issue a 'chmod a+x /path/to/serial-port-json-server' command. Something I didn't have to do in Mojave. Then right-click "open" and confirm that I wanted to bypass Catalina's security warning for running an unrecognized program. Up and running beautifully. Thanks!

closed 2019-12-10 by pastprimitive

nickguycamera, 2020-05-11:
Hi, I realise this thread is a bit out of date but I've been trying to get Chilipeppr working on my mac today, its looks like a great application.  I got as far as finding V1.8 for mac and downloading it but realise this is for Sierra and I am on Mojave.  You have obviously had success in running it on Mojave and I wondered if you could help me out please.  There no longer seems to be a Mac option in the downloads for the latest version.  Which version do I use? Any advice you can offer would be much appreciated.  many thanks!