chilipeppr/serial-port-json-server #65: using in Linux

Gcodeuser, 2019-12-04:
I haven't seen any examples of anyone using serial port json server in Linux.  Has anyone done this?  The readme files suspiciously always have examples of how to install using "Mac/Linux".

chilipeppr, 2019-12-04:
One of the main use cases for SPJS is on Raspberry Pi, so yes, Linux is
full-on supported. There's even a command line switch you can pass to SPJS
that will create the /etc/init.d scripts so SPJS auto starts on boot.

Gcodeuser, 2019-12-05:
Ok, I'm having a bit more success now using the instructions for Raspberry Pi in a Linux Mint environment.  I had to add "sudo" at the start of the first step, where it starts with "wget".
In the second step, shouldn't that read "gzip" instead of "gzid"?

Gcodeuser, 2019-12-05:
For Linux Mint another thing I had to change from the Raspberry Pi instructions is to use “linux_amd64” instead of “linux_arm”