chilipeppr/serial-port-json-server #68: Carriage return instead of linefeed?

scooger, 2020-08-30:
The device I'm interacting with is expecting \r at the end of a command instead of \n. I haven't found anything in the documentation that indicates how to change that behavior. Any ideas?

chilipeppr, 2020-08-30:
What device are you using? The serial port JSON server model is to create a buffer for each device so that it knows how/when each command is processed. It would seem to me you'd need to write a buffer like the other buffers in the source code. Just copy the one that's the closest to what you need.

scooger, 2020-08-31:
I have it working through a web interface (html and javascript) that I am repurposing from the one that controls my Rotel RC-1570 preamp. My devices for this solution are Raspberry Pi 3 B+ to Parasound ZPre3 (stereo preamp) via FTDI USB to RS232. The issue I was experiencing was only for through the serial-port-json-server test interface that appends a linebreak to each command. Fortunately this is not what happens when interacting directly.

closed 2020-08-31 by scooger