chilipeppr/serial-port-json-server #70: LUBUNTU - Syntax Error During Build

PaladinEnabling-Technologies, 2021-07-16:
Hello There,

I'm trying to get this running on lubuntu and when building it and running the following command out of the readme:

`sudo ./serial-port-json-server -createstartupscript`

Then I get the following syntax error:

`./serial-port-json-server: 1: Syntax error: word unexpected (expecting ")")`

Anyone able to guide me in the right direction?

chilipeppr, 2021-07-16:
That startup script is meant for Raspberry Pi OS, so you'd have to go look
at the source code and see what it's doing to see what problem there could
be with Lubuntu that you're using.

PaladinEnabling-Technologies, 2021-07-18:
Hello John,

My mistake, thank you for pointing me in the right direction.

closed 2021-07-18 by PaladinEnabling-Technologies